VoodooI2C/VoodooI2C #540: VooDooI2C breaking sleep wake on MacOS Sonoma RC (as of 21st sep)

Mifaxav, 2023-09-21:
**Describe the bug**
A clear and concise description of what the bug is.

When VoodooI2C is enabled, touchscreen is perfectly fine but it breaks sleep and wake. disabling all the kexts causes the touchscreen to not function but with perfect sleep wake. 

**Did you read the common errors documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Common%20Errors/Common%20Errors

Not yet but I'm pretty sure touchscreen drivers breaking sleep is not common...

**Did you read the troubleshooting documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Troubleshooting/Troubleshooting.

Nothing regarding macOS Sonoma

**Have you searched the issue on Github, Gitter, or Google?**
Please answer yes or no, this is to avoid creating duplicate Github issues.

Yes i have and no results. 

**System Environment**
Please answer:
1. What is your operating system? 

MacOS Sonoma 

2. What version of the drivers (including drivers) are you using?

Latest VoodooI2C drivers

3. What is your hardware configuration?

Im on a Lenovo T480 with a Core i5-8350U processor and a UHD620 GPU

**Troubleshooting Archive**
Please attach the troubleshooting archive (as per https://voodooi2c.github.io/#Troubleshooting/Troubleshooting) and any additional logs or videos demonstrating the problem. Please attach any extra files to Github and *not* anywhere else like Google Drive.

Laptop just reboots(when it is supposed to wake) when voodooI2C kexts are present but disabling all the I2C kexts, sleep wake is perfect.

**Additional context**
Add any other context about the problem here.

N/A

kprinssu, 2023-09-21:
Please attach logs as per the troubleshooting section. Closing this issue until you read and properly fill out the Gthub issue template.

closed 2023-09-21 by kprinssu

Mifaxav, 2023-09-22:
[image]

that's an image reporting an improper sleep wake cycle. This issue only occurs when VooDooI2C is enabled in OpenCore. Disabling all those kexts, i can get perfect sleep wake with S4 hibernation working. I understand this is still in its release candidate form but these changes would most probably be continued into the final release. If needed, i can also test to make sure this issue is fixed.

kprinssu, 2023-09-22:
One again, please read the https://voodooi2c.github.io/#Troubleshooting/Troubleshooting and attach all the logs and IO Registry files. This screenshot is not sufficient nor is it helpful.

This is mandatory and not optional.

kprinssu, 2023-09-23:
This issue is now closed and locked due to the lack of a proper troubleshooting archive.