VoodooI2C/VoodooI2C #552: macOS Sequoia support

dreamwhite, 2024-06-12:
Good morning everyone, I'm opening this issue to keep track of the possible code changes that are required to make VoodooI2C work with the newest release of macOS codename Sequoia.

At the time of writing, VoodooI2C kext v2.8 correctly loads but both in `Interrupt` and `Polling` mode the trackpad doesn't work, totally frozen.

I'm attaching here:

- screenshot of IORegistryExplorer `I2C0.TPD0` device informations:

[image]

- EFI used (with stripped serial numbers + `-lilubetaall` boot-arg: [EFI](https://github.com/dreamwhite/dell-inspiron-5370-hackintosh/tree/master/EFI)

- `AppleIntelLpssI2C` original Sequoia kexts: [AppleIntelLpssI2C.zip](https://github.com/user-attachments/files/15801021/AppleIntelLpssI2C.zip)

Please note that:

1. It's not in my intentions stressing the team
2. with or without `-vi2c-force-polling` boot-arg, the IORegistryExplorer tree of `I2C0.TPD0` is the same as the screenshot
3. in case you need an IORegistryExplorer full dump, is there any ""private"" way of contacting you? I don't wanna spread my serial numbers worldwide :)

Please let me know if you need any other log and additionally how to get them and why not, maybe I can be a tester of new possible versions

Greetings
dreamwhite

dreamwhite, 2024-06-12:
Seems like VoodooI2C issues got fixed after reading this comment by @1Revenger1 on https://github.com/VoodooI2C/VoodooI2CHID/pull/75#issuecomment-2161297712.

I tested it by replacing the bundled `VoodooInput.kext` inside `VoodooI2C.kext/Contents/Plugins`with [this artifact](https://github.com/1Revenger1/VoodooInput/actions/runs/9475484687) both with and without `-vi2c-force-polling` boot-arg, so I'm pretty sure the fix works.

In any case I'll keep this issue opened unless not needed anymore

kprinssu, 2024-06-12:
Thanks for the detailed breakdown. We'll need to update VoodooInput with the updated device identifiers.

narcyzzo, 2024-07-02:
> Seems like VoodooI2C issues got fixed after reading this comment by @1Revenger1 on [VoodooI2C/VoodooI2CHID#75 (comment)](https://github.com/VoodooI2C/VoodooI2CHID/pull/75#issuecomment-2161297712).
> 
> I tested it by replacing the bundled `VoodooInput.kext` inside `VoodooI2C.kext/Contents/Plugins`with [this artifact](https://github.com/1Revenger1/VoodooInput/actions/runs/9475484687) both with and without `-vi2c-force-polling` boot-arg, so I'm pretty sure the fix works.
> 
> In any case I'll keep this issue opened unless not needed anymore

can confirm this works GDX1515 here

kprinssu, 2024-07-02:
Awesome, thanks for also confirming @narcyzzo

kprinssu, 2024-11-05:
@dreamwhite @narcyzzo  Can you please try https://github.com/VoodooI2C/VoodooI2C/releases/tag/v2.9

It should fix the broken trackpads on macOS 15.

AlfCraft07, 2024-11-06:
A new problem has arised: the VoodooInput kext is completely broken on macOS 15.2., as a result completely breaking keyboard and trackpad functionality and essentially turning PCs into paperweights. I guess a fix is needed ASAP, even before final release, as the odds are high the problem will remain.

kprinssu, 2024-11-06:
@AlfCraft07 Instead of just saying that it doesn't work, please post logs. The boot logs should have references to VoodooI2C being loaded. 

The original issue that VoodooInput used an older Magic Trackpad identifier that was removed in macOS 15+. Additionally, your broken keyboard is *not related* to VoodooI2C as this project focus on pointing devices such trackpads, touchpads, touchscreens, etc.

kprinssu, 2024-11-06:
I am closing this as resolved. Please re-open this if the issue persists.

closed 2024-11-06 by kprinssu

AlfCraft07, 2024-11-07:
> @AlfCraft07 The original issue that VoodooInput used an older Magic Trackpad identifier that was removed in macOS 15+. Can you please raise further issues upstream with Acidanthera (https://github.com/acidanthera/bugtracker)?
> 
> I will not be providing support for upstream VoodooInput issues.

I already opened an issue there, but I haven't got any reply yet.

kprinssu, 2024-11-07:
@AlfCraft07 Your issue is an issue with VoodooInput. Please wait for someone there to respond. 

Please avoid spamming here.