VoodooI2C/VoodooI2C #555: X and Y axis not flipping when screen is rotated. (PEN INPUT ONLY)

NotMifa, 2024-09-06:
**Describe the bug**
I have a lenovo X1 yoga gen 3 and when I use it in potrait mode, the lenovo pen's input stays the same.. (What I mean to say is the x and y axis dont flip once the screen is rotated). Gestures and finger touch works fine, its just the pen thats having the issue

**Did you read the common errors documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Common%20Errors/Common%20Errors

Yes

**Did you read the troubleshooting documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Troubleshooting/Troubleshooting.

Yes

**Have you searched the issue on Github, Gitter, or Google?**
Please answer yes or no, this is to avoid creating duplicate Github issues.

Yes, nothing relating to my issue

**System Environment**
Please answer:
1. What is your operating system? 

Macos Sonoma 14.2.1

3. What version of the drivers (including drivers) are you using?

Using Opencore 1.0, and a compile of the latest master(voodoo I2C), latest versions of voodooinput and voodooPS2

5. What is your hardware configuration?

Laptop Model: Thinkpad X1 yoga gen 3
CPU: Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz (UHD620 GPU)
16 GB of ram
Opencore bootloader 1.0
Using voodooI2C for a touchscreen

**Troubleshooting Archive**
Please attach the troubleshooting archive (as per https://voodooi2c.github.io/#Troubleshooting/Troubleshooting) and any additional logs or videos demonstrating the problem. Please attach any extra files to Github and *not* anywhere else like Google Drive.

[MacbookPro.ioreg.zip](https://github.com/user-attachments/files/16902709/MacbookPro.ioreg.zip)
[voodooI2C_issue.txt](https://github.com/user-attachments/files/16902732/voodooI2C_issue.txt)
[system.log.txt](https://github.com/user-attachments/files/16902749/system.log.txt)
[Yash's HackBook Pro.acpi.zip](https://github.com/user-attachments/files/16902799/Yash.s.HackBook.Pro.acpi.zip)

**Additional context**
N/A

P.S: Will send a video of the issue once I get home. 

if any other files are required, will be more than happy to share them..!

kprinssu, 2024-09-06:
Pen support has always been flaky. I am sorry but I will not be fixing this issue.

closed 2024-09-06 by kprinssu

NotMifa, 2024-09-06:
Hey, seems like the latest master has solved this issue, was using an older compiled version of the kext. VoodooI2C 2.9 or 2.8.5? Or, could you guide me a bit though the process of building the kexts?

kprinssu, 2024-09-07:
The README contains a link to the documentation which has build instructions.  I will not be providing support on how to build the kexts.